opentechinstitute/commotion-router #161: New Airgrid M5's unable to be flashed with Commotion

Protagonistics, 2015-01-18:
Meta Mesh has recently purchased a number of Airgrid M5's and has attempted to flash v1.1 on them as it has done in the past but has been unable to. This is a critical problem and essentially stalls all long-distance expansion for everyone using Commotion.

After verifying the MD5 hash from a fresh download of Commotion for Airgrid, we are unable to upload via gui the firmware, receiving the error -6 in this screenshot:

[image]

After doing a bit of research, it appears that Ubnt has changed the file format of its images specifically for the Airgrid M5: https://community.ubnt.com/t5/airMAX-Frequently-Asked/airMax-Choosing-the-right-firmware-file-for-your-device-upgrade/ta-p/907346

This makes sense with what we have seen. We have attempted to upload v1.1, rc1, a nightly build, a vanilla openwrt build, and have attempted this from gui and in tftp mode. the check fails in all builds and the only kind of upgrade that can be made is upgrades to the UBNT firmware from their site.

This is a huge issue! I hope we can address this immediately. Meta Mesh is at your disposal to assist in testing this.

Protagonistics, 2015-01-18:
edit, the screenshot is from a sysupgrade image which gives a -4 error which is actually normal I believe. The issue we are facing is -6, I just uploaded the wrong screen shot.

jheretic, 2015-01-20:
Just to keep this updated in Github:

As per discussion in #commotion, it does look like Ubiquiti's started quietly rolling out XW-series Airgrids, which are not currently supported by OpenWRT. The XW Nanostations are supported in Barrier Breaker, and the XW Locos in trunk, but there are no patches yet for Bullet-style devices like the Airgrids. We're currently evaluating whether we can support these devices with Commotion and how best to do so.