opentechinstitute/commotion-router #164: Give router image files human-readable names

andygunn, 2015-02-10:
During our last workshop, the very long file names for the Commotion images was a source of confusion. I understand there is a lot of information encoded in the name of the image file, and that information is important, but 99% of people can't decode that information anyways. A more useful name would include:
* The name of the distribution of openwrt: Commotion
* The version of Commotion: 1.1 or 1.2, etc
* The manufacturer: TP-Link, Ubiquiti, etc
* The router model: WDR3600-v1 or Rocket-M-XM or Rocket-M-XW, etc
* Whether the file is an installation or upgrade file.

The existing filename 
**openwrt-ar71xx-generic-ubnt-bullet-m-squashfs-factory.bin**
would change to:
**Commotion-1.1-Ubiquiti-Bullet-M-install.bin**
It would also be necessary to create copies of this file that were identical, but had a different name, such as:
**Commotion-1.1-Ubiquiti-PicoStation-M-install.bin**, **Commotion-1.1-Ubiquiti-AirGrid-M-install.bin**

This strips the information most people aren't familiar with or able to decode - ar71xx, generic, squashfs.

This would eliminate the confusion between the different types of files, and what version has been downloaded (a problem other networks have encountered in the past, when dealing with versions of Commotion over several years).

dismantl, 2015-02-10:
We've changed the output filenames slightly in the new build system, however we thought that anyone compiling their own images would be able to parse the filenames, and would need the extra information contained in them.

We thought it would be better to rename the files on the download page of the website to something easy, like you suggest. What do you think of that idea?

andygunn, 2015-02-10:
That would be fine as well - I was contemplating putting this in the commotion-docs repo, but I wasn't sure where along the path files were named and signed, so I put it here.

I understand folks that are building their own might be able to parse the names, so that makes sense. Most folks who encounter the filenames for the first time just screech to a halt!