opentechinstitute/commotion-router #166: TX offset not being added to TX power value

andygunn, 2015-02-25:
On PicoStation M2 units and perhaps other Ubiquiti equipment, the value for TX power offset (i.e. the lowest power output of the unit) is not being added in. This causes the power output to display as 16dBm, whereas it should be 28dBm (the TX power offset is 12dBm, the lowest the PicoStation M2 can go).

At least in the LuCi menus, this should be summed in to show the true output level.

Reported in [support ticket 1240](https://projects.opentechinstitute.org/issues/1240).

andygunn, 2015-04-27:
@jheretic @dismantl - Any chance this could get resolved for 1.2? I am noticing the same bug occur in NanoStation and Rocket units in 1.2...

westbywest, 2015-04-27:
How are you checking the radio's current tx power level?  E.g. with
iwlist?  Or trying to measure the actual signal strength observed vs. tx
dBm value set?

If the tx offset is not being added in, i.e. radio always underpowered, you
may have to bug the OpenWRT folks (or wireless-testing) folks upstream to
get a resolution.