opentechinstitute/commotion-router #176: "http://thisnode" alias not working on Windows clients

andygunn, 2015-06-11:
There have been several reports of the http://thisnode alias not working on Windows clients - and during an international workshop I saw this first hand. It is repeatable - the alias will not work, only going to the .1 router IP address on the LAN subnet.

dismantl, 2015-06-11:
Isn't resolving to the 10.x.x.1 IP on the LAN subnet the expected behavior?

andygunn, 2015-06-11:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

All of our documentation references the http://thisnode alias. Most
people do not know how to look up their IP address and intuit the
router address from that.

> Isn't resolving to the 10.x.x.1 IP on the LAN subnet the expected
> behavior?
> 
> <https://github.com/opentechinstitute/commotion-router/issues/176#issu
ecomment-111184013>.
- -- 
Andy Gunn, Field Engineer
Open Technology Institute, New America
andygunn@opentechinstitute.org | 202-596-3484
PGP: F1D2 CD5E 9F15 EEB0 232A 1EFA EEDC DC5C F1D5 653C
-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2.0.22 (GNU/Linux)

iQEcBAEBAgAGBQJVebDLAAoJEO7c3Fzx1WU8LXwH/iVyI6NVFAdMsGYwp7sz6tal
b5jk3+3GDuBs3e09CwqhtZIxR5/mjehIqKeNAxCrJio4ufes9JsbD/zDQbAJTRT3
cZNGxJjnj8Oe/JyAWNArDS+9Zg55t4SnsfeI/B6tw2PPWeroo84zR/1DoVNnnjWz
WlXFIQnSxV2Z6pjFmTVqVYH2QSHDnMdF/Ung7bPlhYPi2oPKtiEkiZknZH4fzm2s
VGeYu4JS870JhtZ1IDdRHCvP1pC3lNtHGYGEKr+TWyjcTAefi6CXZy1LfhcU3y0S
5fDNBkcGx2hEM8SP+A1FOThXc9okrAJqlaMomWPVxvinOx1q4JWj9/pdsFd6RIs=
=Fb4A
-----END PGP SIGNATURE-----

dismantl, 2015-06-11:
Sorry, I misunderstood what you said initially. On the Windows client, can you see what DNS server it is configured to use?

You can also manually resolve the hostname with Powershell and see what that gives you: `[System.Net.Dns]::GetHostAddresses("thisnode")`

andygunn, 2015-06-11:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

This is just from reports. I believe the thisnode address resolves to
the zeroconf IP, so that could be the issue. Next time I am on a
Windows computer I can check the resolution.

> Sorry, I misunderstood what you said initially. On the Windows
> client, can you see what DNS server it is configured to use?
>
> You can also manually resolve the hostname with Powershell and see
> what [...]
ecomment-111185760>.
- -- 
Andy Gunn, Field Engineer
Open Technology Institute, New America
andygunn@opentechinstitute.org | 202-596-3484
PGP: F1D2 CD5E 9F15 EEB0 232A 1EFA EEDC DC5C F1D5 653C
-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2.0.22 (GNU/Linux)

iQEcBAEBAgAGBQJVebIJAAoJEO7c3Fzx1WU8AnEIAJyVSl5YUz9Mvc8V0PlIldkQ
SI8pSrMWLSpnMTDRFnVrE3AklkruvUUV4t+/EscGjfxCNxRsoPizrT9Nl9ITXn6Q
rynC+O/oqmAtStMswQ98AS24aUV0Vj7dog5XqdeateS3yrYEZNNQoYQpx0nypW5i
vFuMg9QiAy59tJ4BfVYZ0KllbR8ZZT+77rZ+ScILJ5Toq3k+MR3GAgvslv17R8qW
qFeLEVwqDJV6jcZJOzQ1ug1q7H1gNL5TYDyVyE4D3Tx6eBYgaXLxrfd355FE9hlS
kHSq5JA1GrosmTGqNT8taIqsmFMxelvy3fJENbKuoDLUR2kLiQqhpOZh1C6Wsv4=
=3UIy
-----END PGP SIGNATURE-----